hardbyte/python-can #1777: Single millisecond send resolution with python-can, Vector

ack-kullum, 2024-05-12:
python version: 3.9.13, python-can version: 4.0.0

I am trying to send a CAN message every millisecond using python-can to access Vector's CANalyzer software. Using the code below, I would expect to see a message every millisecond:
```
import can
import time
import logging

bus = can.interface.Bus(bustype="vector", channel=0, bitrate=500000, app_name='CANalyzer')

def periodic_send(msg, bus):
    task = bus.send_periodic(msg, 0.001)
    assert isinstance(task, can.CyclicSendTaskABC)

    return task

tasks = []

msg = can.Message(arbitration_id=0xCE, data=[1, 1, 1, 1], is_extended_id=False)
tasks.append(periodic_send(msg,bus))

time.sleep(200)
```

However, when I look at the signal in CANalyzer, it shows the message being sent every 15ms or so (blue measurement cursor is first sample, orange is second):
[image]

I have also tried managing timing directly and have similarly been unable to achieve single millisecond intervals. Is this possible to achieve with python-can?

danielhrisca, 2024-05-21:
Does it make any difference if you patch `time` before importing `can`

```python

import sys

if sys.platform == "win32":
    import time

    from win_precise_time import sleep as wpsleep
    from win_precise_time import time as wptime

    time.sleep = wpsleep
    time.time = wptime

import can

```

see https://pypi.org/project/win-precise-time/

pierreluctg, 2024-05-21:
On Windows, which I assume you are since you are using Vector, python-can not using `time.sleep` in most setup.
See https://github.com/hardbyte/python-can/blob/main/can/broadcastmanager.py#L327

On of the reason to use `win32event` is exactly for the reason that you are raising. `time.sleep` (prior to Python 3.12) has a resolution of 15.6 ms.

Now you should try to figure out why in your setup `USE_WINDOWS_EVENTS` is `False`
https://github.com/hardbyte/python-can/blob/main/can/broadcastmanager.py#L23