hardbyte/python-can #1786: Crash in ASCReader

DuyPhuong96, 2024-06-04:
### Describe the bug
I'm a new in CANoe and I'm trying to read an asc file by ASCReader.
This asc file is automatically generated and it has a special message at the beginning: `0.000000 CANFD   Start of measurement`
ASCReader crashed while trying to parse this message.

asc.py crash source:
```
    def _process_fd_can_frame(self, line: str, msg_kwargs: Dict[str, Any]) -> Message:
        channel, direction, rest_of_message = line.split(None, 2)
        # See ASCWriter
        msg_kwargs["channel"] = int(channel) - 1
        msg_kwargs["is_rx"] = direction == "Rx"
```

Error:
```
  File "C:\Users\AppData\Local\Programs\Python\Python312\Lib\site-packages\can\io\asc.py", line 211, in _process_fd_can_frame
    msg_kwargs["channel"] = int(channel) - 1
                            ^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'Start'
```
### To Reproduce
Prepare an asc file with `0.000000 CANFD   Start of measurement` message
```
date Fri May 05 06:24:08.927000 PM 2024
base hex  timestamps absolute
internal events logged
Begin Triggerblock Fri May 05 06:24:08.846 PM 2024
   0.000000 CANFD   Start of measurement
   0.000000 CANFD   1 Rx        4B0                                   1 0 8 8 00 00 00 00 00 00 00 00   0  0   323040 0 0 0 0 0
```
Read this asc file by ASCReader:
```
import can

file_path = "example.asc"

with can.ASCReader(file_path) as log:
    for message in log:
        print(message)
```

### Expected behavior
I know the format of message is not correct.
But I think the lib should ignore the wrong format message like that and continue to reader other.

I'm sorry if my thinking is wrong.
Thank you!
### Additional context

OS and version:
Python version:
python-can version:
python-can interface/s (if applicable):

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

Tian-Jionglu, 2024-06-11:
Is this log can be read by CANoe?

As far as I know, you log should be 
```
0.000000 Start of measurement
```
It should be a flag of Start. There should be no CANFD flag.

RitheeshBaradwaj, 2024-07-08:
Hi, as per my understanding, the line:

```
0.000000 CANFD Start of measurement
```

appears to be incorrect. It seems to be a combination of two different conventions:

1. **CANoe Start of Measurement:** typically uses a line like `0.000000 Start of measurement` at the beginning of a log file to indicate its start. There's no "CANFD" flag in this case.

2. **CANFD Indication:** The "CANFD" flag is used in ASC files to distinguish CANFD frames from standard CAN frames. However, this flag should only be present in the channel column for actual CANFD messages, not for a log start indicator.

To robustly handle this, we can implement checks to detect and skip special messages like "Start of measurement" before processing CANFD frames

I made the following change in the `__iter__` method to handle the "Start of measurement" lines separately and was able to verify this fixes the issue

https://github.com/hardbyte/python-can/pull/1811

PR #1811 (2024-07-26): Fix ASCReader Crash on "Start of Measurement" Line
### Summary

This PR addresses an issue where the ASCReader crashes when encountering the line `0.000000 CANFD Start of measurement` in ASC log files. This line is a combination of two different conventions and should not contain the "CANFD" flag.

### Changes

- Added logic in the `__iter__` method of `ASCReader` to skip lines matching the regex pattern for "Start of measurement".
- This ensures that such special indicator lines are handled separately and do not cause parsing errors.

Fixes #1786.
closed 2024-07-26 by hardbyte