hardbyte/python-can #1815: can.player infinite loop support

KhazAkar, 2024-07-11:
### Is your feature request related to a problem? Please describe.
I have blf recording which I would want to play indefinitely, without the need to restart the replay manually.

### Describe the solution you'd like
Something like `--loop` argument.

### Describe alternatives you've considered
Writing custom can replay script based on original one.

Tbruno25, 2024-08-02:
Hmm, this seems like a somewhat niche and specific use case as I've never heard of a "circular log" before. There will be a slight delay on every iteration, but could this be easily achieved by executing `can.player` within a bash loop?

KhazAkar, 2024-08-02:
It's like trying to simulate something like CANoe replay block which plays BLF file indefinitely. Bash loop might work in that case, but having something working on linux and windows (native, not WSL) would be best :P 
If that doesn't fit in the scope - that's fine :)