hardbyte/python-can #1860: Add support for complex channel names in ASC files

rettenbs, 2024-09-17:
### Is your feature request related to a problem? Please describe.
We encountered an ASC file that contains channels names like `Li`:
```
  42.017364 Li 14              Rx     8 2a 79 00 00 00 80 c0 ff  checksum = 07   header time =  40, full time = 135  SOF =   42.010348   BR = 19200  break = 942203 109234   EOH =   42.012418    EOB =   42.012979    42.013527    42.014075    42.014623    42.015172    42.015720    42.016268    42.016816    sim = 0    EOF =   42.017364   RBR = 19157  HBR = 19202.640363  HSO = 22785     RSO = 22837     CSM = enhanced
```
The current ASC reader fails to parse these files, i.e. it skips these messages because channel has to be an int.

### Describe the solution you'd like
Would it be possible to enhance the ASC reader to deal with the complex channel names?

zariiii9003, 2024-09-17:
That's a LIN channel, not a CAN channel.

rettenbs, 2024-09-17:
This makes a lot of sense ... Thank you very much.

Do you think it is feasible to support LIN channels as well for ASC files? I know this is CAN library but it is the best reader for ASC files for Python.

zariiii9003, 2024-09-18:
But what are we supposed to do with these lines? It does not make sense to return `can.Message` instances for LIN messages.

rettenbs, 2024-09-20:
I see. We would need a `lin.Message`. Probably beyond the scope of this library.