hardbyte/python-can #1873: To integrate a TOSUN CAN device

sy950915, 2024-10-15:
Hello, I am an employee of Shanghai Tongxing Intelligent Technology Co., Ltd. We are interested in integrating our company's device into Python-can. Currently, many users are requesting the ability to use our device directly within Python-can. At present, we achieve this by installing Python-can and then adding our own files into the Python-can directory to enable this functionality. However, this approach is not as seamless as having our device fully integrated into Python-can, and we are looking for a more convenient solution.

zariiii9003, 2024-10-15:
Hi, you can create and publish your own package, which python-cans entry point as described [here](https://python-can.readthedocs.io/en/main/plugin-interface.html). We will add your package as an extra dependency and mention it in our documentation.

sy950915, 2024-10-15:
@zariiii9003 Hello, at the moment, our approach requires users to first install Python-can and then configure it manually in order to use our device. We would like to discuss the possibility of directly integrating our device into Python-can, so that users can install Python-can via pip and use our device without any additional configuration—similar to how Vector devices are supported.

zariiii9003, 2024-10-15:
Your users would install your package as an extra dependency with `pip install python-can[tosun]`. No additional configuration necessary.

sy950915, 2024-10-15:
@zariiii9003 
Alright, thank you very much. I will take some time to research this further. I greatly appreciate your response.

luojiaaoo, 2024-10-21:
I've manually added your compatible python file to python-can @sy950915 ,There were a few small problem. I changed some code manually for TOSUN.
I hope the next version of python-can, tosun will be officially supported. Python-can is the most popular library in python, so focus on it, (- v -)