hardbyte/python-can #1877: kvaser timestamps broken since 4.4.1

johanb321, 2024-10-21:
### Describe the bug

Timestamps from kvaser messages are errouneous, due to the timestamp offset applied is incorrectly initialized in init function. 
Looks like the root cause is that the calls to canBusOn were moved in version 4.4.1 to now be done after the calls to kvReadTimer used to initialize the timestamp offset, this is causing the kvReadTimer to return a value that is not right (a way much bigger value than what is returned as timestamps on messages received later!). 

### To Reproduce

Check timestamps on received messages from _recv_internal function and compare them with the value of time.time() when the function returned, it should be close to zero but is not. 

### Expected behavior

Timestamps should be accurate and in sync with values returned by time.time()

### Additional context

OS and version: Windows 10
Python version: 3.11.7
python-can version: 4.4.2
python-can interface/s (if applicable): Kvaser Leaf light v2

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

zariiii9003, 2024-10-21:
Thank you for reporting this. Can you confirm, that #1878 solves the issue?

johanb321, 2024-10-21:
Tried it out and it works just fine!

PR #1878 (2024-10-25): Fix Kvaser timestamp
Fixes #1877