hardbyte/python-can #1883: FEATURE REQUEST: Support for MDF4 CAN bus logging files [PR ready]

MatinF, 2024-10-30:
### Is your feature request related to a problem? Please describe.

The current MDF4 reader in the `python-can` project is only able to load MDF4 files generated by the MDF4 writer in the project, or MDF4 files having the exact same internal structure. This severely limits the number of MDF4 files with CAN data which can be loaded into the `python-can` ecosystem for analysis and playback.

### Describe the solution you'd like

An expansion of the MDF4 reader capabilities to support MDF4 CAN bus logging files in general. I.e. instead of always expecting CAN DataFrames to be group 0, allow these to be any group, and even span multiple groups. The `asammdf` package the reader/writer leverages for the heavy lifting behind the scenes already supports loading of more complex files/other structures.

We can provide a PR for expanding the MDF4 readers capabilities to support more generalized CAN bus logging files (in fact we have already prepared and tested one).

Please let us know whether this is something you are open to, then we will formally prepare the PR.

MatinF, 2024-11-01:
As an update, we have now made a PR for this:
https://github.com/hardbyte/python-can/pull/1884

johnMediaOnFire, 2025-01-20:
I think you can close this as a duplicate of #1884.

closed 2025-01-21 by MatinF