hardbyte/python-can #1885: Trouble Applying Filters to Bus with Vector Interface

PyCoder9595, 2024-11-01:
Attempting to filter bus to receive single PGN with vector interface. Filter is applied but still receiving other IDs when calling recv(). Not sure if issue with filter or bus declaration. 

[image]
[image]

zariiii9003, 2024-11-03:
Try this:
```python3
filters = [{'can_id': 0xfff, 'can_mask': 0xfff, 'extended': False}, {'can_id': 0x18daf9aa, 'can_mask': 0x1fffffff, 'extended': True}]
```

PyCoder9595, 2024-11-04:
Currently this is being tested with Vector CAN case 1640A.

phospho666, 2024-11-06:
Hi, I add the same issue with an MCP2515 controler, in my case, the filter was applied after the receive buffer so when reading, I was receiving old messages stored in the buffer before the filters were applied.