hardbyte/python-can #1899: Function canWrite failed - Transmit buffer overflow

AlanCodeSpace, 2024-11-25:
### Describe the bug

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/can/broadcastmanager.py", line 308, in _run
    self.bus.send(self.messages[msg_index])
  File "/usr/local/lib/python3.8/dist-packages/can/interfaces/kvaser/canlib.py", line 690, in send
    canWrite(
  File "/usr/local/lib/python3.8/dist-packages/can/interfaces/kvaser/canlib.py", line 103, in __check_status_operation
    raise CANLIBOperationError(function, result, arguments)
can.interfaces.kvaser.canlib.CANLIBOperationError: Function canWrite failed - Transmit buffer overflow [Error Code -13]

### To Reproduce

### Expected behavior

### Additional context

OS and version:
Python version:
python-can version:
python-can interface/s (if applicable):

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

luojiaaoo, 2024-11-25:
Asynchronous sending is too fast. The hardware cache overflows
You should add timeout in send
or change python-can kvaser code, force to canWriteSync per 128 frame

AlanCodeSpace, 2024-11-25:
> canWriteSync

I am using the send_periodic() interface, it seems that there is no way to set the timeout?
Or how to set this timeout?

luojiaaoo, 2024-11-27:
```
 def _run(self) -> None:
        msg_index = 0
        msg_due_time_ns = time.perf_counter_ns()

        if USE_WINDOWS_EVENTS:
            # Make sure the timer is non-signaled before entering the loop
            win32event.WaitForSingleObject(self.event.handle, 0)

        while not self.stopped:
            if self.end_time is not None and time.perf_counter() >= self.end_time:
                break

            # Prevent calling bus.send from multiple threads
            with self.send_lock:
                try:
                    if self.modifier_callback is not None:
                        self.modifier_callback(self.messages[msg_index])
                    ****************** here *************
                    self.bus.send(self.messages[msg_index])
                    ****************** here *************
                except Exception as exc:  # pylint: disable=broad-except
                    log.exception(exc)

                    # stop if `on_error` callback was not given
                    if self.on_error is None:
                        self.stop()
                        raise exc

                    # stop if `on_error` returns False
                    if not self.on_error(exc):
                        self.stop()
                        break

            if not USE_WINDOWS_EVENTS:
                msg_due_time_ns += self.period_ns

            msg_index = (msg_index + 1) % len(self.messages)

            if USE_WINDOWS_EVENTS:
                win32event.WaitForSingleObject(
                    self.event.handle,
                    win32event.INFINITE,
                )
            else:
                # Compensate for the time it takes to send the message
                delay_ns = msg_due_time_ns - time.perf_counter_ns()
                if delay_ns > 0:
                    time.sleep(delay_ns / NANOSECONDS_IN_SECOND)
```
It is no way to set the timeout
You can use "while" to archive this send_periodic nstead of calling send_periodic()
Or you can change code of python-can

closed 2024-12-03 by AlanCodeSpace