hardbyte/python-can #1928: ValueCan4 Send returns timeout exception but message make its through

jag213, 2025-03-06:
Hello,

I am using a ValueCan4 to communicate with a TI microcontroller. I've implemented the short code below to test a message transmit and get a timeout exception when I run the code as is. However, I am seeing the message show up on the DSP so the message seems to be getting through. Through dumb trial and error I have also found that if I uncomment the "notifier =" line I don't get exception and the message makes it through as well. 

Console when notifier line commented out 

Message NOT sent: Transmit timeout

Console when notifier line left in 

Timestamp: 573622063.141707    ID: 00000300    X Tx                DL:  8    01 00 00 00 00 00 00 00     Channel: 1
Send success

Timestamp: 573622063.141715    ID:      000    S Rx E              DL:  3    14 00 82                    Channel: 1

I've tried power cycling the DSP, flushing the valuecan buffer, unplugging/reconnecting the ValueCan but still get the exception if the "notifier" line is removed. Don't know if this is a bug or I am just doing something dumb.

Thanks

Versions
python                    3.9.19               
python-can                4.3.1  

```python
import can

bus = can.Bus(bustype='neovi', serial='V13822', channel=1)

#Start the notifier to print out bus messages
# notifier = can.Notifier(bus, [can.Logger("recorded.log"), can.Printer()])

msg = can.Message(arbitration_id=0x300,data=[1, 0, 0, 0, 0, 0, 0, 0], is_extended_id=True)
try:
    bus.send(msg, timeout=0.5)
    print('\nSend success')
except can.CanError as e:
    print("\nMessage NOT sent:", e)

bus.shutdown()
```
</details>

zariiii9003, 2025-03-06:
I found this code comment in the neovi bus:

```
        # If timeout is set, wait for ACK
        # This requires a notifier for the bus or
        # some other thread calling recv periodically
```
Could you try it?

jag213, 2025-03-06:
@zariiii9003 @pierreluctg Thanks for the responses. Adding the notifier line solves the issue (setting timeout=0 also solves it). 

I originally had the timeout in a different application using a GUI interface and found the GUI hung up if the send failed and timeout was not set.

closed 2025-03-06 by jag213